nordicsemi/Android-Scanner-Compat-Library #110: problem with build,, Caused by: groovy.lang.MissingPropertyException: Could not get unknown property 'sonatypeStagingProfileId' for object of type io.github.gradlenexus.publishplugin.NexusRepository.

Dev-Damon, 2021-09-29:
Caused by: groovy.lang.MissingPropertyException: Could not get unknown property 'sonatypeStagingProfileId' for object of type io.github.gradlenexus.publishplugin.NexusRepository.

couldn't find the solution

any help??

thanks in advanced

philips77, 2021-09-29:
Hi,
For now, you may comment out last lines in *build.gradle* files, which import release scripts. 
I'll do a release today that fixes #106 and your issue.

Dev-Damon, 2021-09-30:
thanks for quick answer!
I sorry but it still doen't work,
I pulled develop branch than you just commited yesterday, but it's still same... 
does nobody run into this situation? ...
i donno what to do

philips77, 2021-09-30:
Yes, I know. I've fixed the build script internally, but want to test the other changes before i push it 

Question, why do you want to open this project directly, not include it from maven central?

Dev-Damon, 2021-10-01:
I wasn't aware of that.
is it just made for runn in maven central? not run directly??
I'm looking for example app which can run directly. I think this is not the one I'm looking for. should find other one.
Thanks for help! I really appreciate

philips77, 2021-10-01:
You may use [nRF Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky). I'll update it, hopefully today, as it also needs some fixing.

philips77, 2021-10-01:
> is it just made for runn in maven central? not run directly??

Yes, in this repo we just have the library, with no sample app.