nordicsemi/Android-Scanner-Compat-Library #122: Bluetooth 5 Advertisment extensions device not scannable

trix0, 2022-07-06:
Hi, 
We have changed to Advertising type: Bluetooth 5 Advertising extension
The device does not show up in scan result.(Works in nrF connect App) 
Is this type supported by the scanner? 

Thanks,
T

philips77, 2022-07-10:
Hi,
Yes, it's supported on Android 8+, i think. You need to call `setLegacy(false)` to make it work.

closed 2022-07-10 by philips77

trix0, 2022-07-12:
Device: Lenovo TB-X505L 
Android version:10(29)
This is how my startScan looks like. 

   ```
            List<ScanFilter> filters = new ArrayList<>();
            scanSettings=new ScanSettings.Builder().setLegacy(false).setScanMode(ScanSettings.SCAN_MODE_LOW_POWER).build();
            scanner.startScan(filters,scanSettings,scanCallback);
```

Any idea what I could try?

philips77, 2022-07-12:
Try with LOW_LATENCY mode.
You may also try changing scan mode to low power in nRF Connect in Settings / Scanner, and check if it doesn't work as well.