nordicsemi/Android-Scanner-Compat-Library #123: Android 13 Support

junebeans, 2022-07-26:
Hello, I'm one of the software engineers using your Bluetooth libraries. First of all, I really appreciate the efforts to resolve a tremendous amount of compatibility issues across different levels of operating system versions.

I was wondering if there are any ongoing development regarding compatibility with Android 13 as there were some level API level changes coming in. This might also affect your [ble library](https://github.com/NordicSemiconductor/Android-BLE-Library).

I noticed that there were some changes as the manufacturer data was different from what we have and was investigating where the issue is coming from. We investigated and we're suspecting that the changes from the operating system on Android 13 are causing the compatibility issue.

- https://developer.android.com/sdk/api_diff/33/changes/pkg_android.bluetooth
- https://developer.android.com/sdk/api_diff/33/changes/pkg_android.bluetooth.le

Is there any ongoing effort, maybe our team can try it out and see if it is resolved and find potential issues?

Thanks.

philips77, 2022-07-27:
Hi, thank you for bringing my attention to it. Last time I checked the API changes, there was nothing related to scanning, but seems like now there is a lot, related to data types, for example. This should be addressed in this library. I'll try to find some time to implement it.

As to the BLE Library, the only changes are `writeCharacteristic` and `writeDescriptor` and `notifyCharacteristicChange`, which are handled inside the app. I'll migrate to use the new API, but there should be no API changes for the user. Still, new version is expected.

junebeans, 2022-12-06:
@philips77  I'm not sure if you'd want to close this open issue. However, I updated the library ble library v2.6.0 alpha 04 (from 2.4.1) to test on android 13 devices and it worked. I will keep my eye on the later stable releases but wanted to give an update as I've opened this thread.

Thanks

philips77, 2022-12-08:
Thanks for the reminder. I think the migration to Android 13 is complete and there's nothing in this library to be changed, perhaps except `targetSdkVersion`, but that's not important. I'll close the issue.

closed 2022-12-08 by philips77