nordicsemi/Android-Scanner-Compat-Library #129: NoClassDefFoundError: Failed resolution

khynantoque, 2022-11-07:
I am making a unity plugin and I want to acquire rssi values from my ble but my app didn't launch, and I got this error in my logcat.
`FATAL EXCEPTION: main
                                                                                                    Process: com.khynsoft.EyeNav, PID: 8094
                                                                                                    java.lang.NoClassDefFoundError: Failed resolution of: Lno/nordicsemi/android/support/v18/scanner/BluetoothLeScannerCompat;
                                                                                                    	at com.khynsoft.ble.PluginActivity.onCreate(PluginActivity.java:26)
                                                                                                    	at android.app.Activity.performCreate(Activity.java:8214)
                                                                                                    	at android.app.Activity.performCreate(Activity.java:8202)
                                                                                                    	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1320)
                                                                                                    	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:4033)
                                                                                                    	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4247)
                                                                                                    	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:91)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:149)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:103)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2613)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:110)
                                                                                                    	at android.os.Looper.loop(Looper.java:219)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8668)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109)
                                                                                                    Caused by: java.lang.ClassNotFoundException: no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat
                                                                                                    	at com.khynsoft.ble.PluginActivity.onCreate(PluginActivity.java:26) 
                                                                                                    	at android.app.Activity.performCreate(Activity.java:8214) 
                                                                                                    	at android.app.Activity.performCreate(Activity.java:8202) 
                                                                                                    	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1320) 
                                                                                                    	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:4033) 
                                                                                                    	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4247) 
                                                                                                    	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:91) 
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:149) 
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:103) 
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2613) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:110) 
                                                                                                    	at android.os.Looper.loop(Looper.java:219) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8668) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109) `
my code:
`BluetoothLeScannerCompat scanner = BluetoothLeScannerCompat.getScanner();
        ScanSettings settings = new ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(5000)
                .setUseHardwareBatchingIfSupported(true)
                .build();
        ScanCallback scanCallback = new ScanCallback() {
            @Override
            public void onScanResult(int callbackType, @NonNull ScanResult result) {
                super.onScanResult(callbackType, result);
                Toast.makeText(PluginActivity.this, result.toString(), Toast.LENGTH_SHORT).show();
            }

            @Override
            public void onBatchScanResults(@NonNull List<ScanResult> results) {
                for (ScanResult res : results) {
                    if (ActivityCompat.checkSelfPermission(PluginActivity.this, Manifest.permission.BLUETOOTH_CONNECT) != PackageManager.PERMISSION_GRANTED) {
                        Toast.makeText(PluginActivity.this, "Permission is disabled!", Toast.LENGTH_SHORT).show();
                        return;
                    }
                    Toast.makeText(
                            PluginActivity.this,
                            String.format(Locale.ENGLISH,"Device: %s RSSI %d", res.getDevice().getName(), res.getRssi()),
                            Toast.LENGTH_SHORT
                    ).show();
                }
            }

            @Override
            public void onScanFailed(int errorCode) {
                super.onScanFailed(errorCode);
                Toast.makeText(PluginActivity.this, "Scan is bulokens!", Toast.LENGTH_SHORT).show();
            }
        };
        List<ScanFilter> filters = new ArrayList<>();
        filters.add(new ScanFilter.Builder().setDeviceName("KunwareBLE1").build());
        scanner.startScan(filters, settings, scanCallback);`

philips77, 2022-11-07:
Hi,
The error indicates a dependency issue. Check if the library is accessible from your code, Proguard is not removing it, or you're not using `debugImplementation` in gradle configuration.

khynantoque, 2022-11-07:
The library is accessible in my code
```
package com.khynsoft.ble;

import android.Manifest;
import android.content.pm.PackageManager;
import android.os.Bundle;
import android.widget.Toast;

import androidx.annotation.NonNull;
import androidx.core.app.ActivityCompat;

import java.util.ArrayList;
import java.util.List;
import java.util.Locale;

import no.nordicsemi.android.support.v18.scanner.BluetoothLeScannerCompat;
import no.nordicsemi.android.support.v18.scanner.ScanCallback;
import no.nordicsemi.android.support.v18.scanner.ScanFilter;
import no.nordicsemi.android.support.v18.scanner.ScanResult;
import no.nordicsemi.android.support.v18.scanner.ScanSettings;

public class PluginActivity extends UnityPlayerActivity{
    BluetoothLeScannerCompat scanner;
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        Toast.makeText(this, "Yawa!", Toast.LENGTH_SHORT).show();

        scanner = BluetoothLeScannerCompat.getScanner();
        ScanSettings settings = new ScanSettings.Builder()
                .setLegacy(false)
                .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                .setReportDelay(5000)
                .setUseHardwareBatchingIfSupported(true)
                .build();
        ScanCallback scanCallback = new ScanCallback() {
            @Override
            public void onScanResult(int callbackType, @NonNull ScanResult result) {
                super.onScanResult(callbackType, result);
                Toast.makeText(PluginActivity.this, result.toString(), Toast.LENGTH_SHORT).show();
            }

            @Override
            public void onBatchScanResults(@NonNull List<ScanResult> results) {
                for (ScanResult res : results) {
                    if (ActivityCompat.checkSelfPermission(PluginActivity.this, Manifest.permission.BLUETOOTH_CONNECT) != PackageManager.PERMISSION_GRANTED) {
                        Toast.makeText(PluginActivity.this, "Permission is disabled!", Toast.LENGTH_SHORT).show();
                        return;
                    }
                    Toast.makeText(
                            PluginActivity.this,
                            String.format(Locale.ENGLISH,"Device: %s RSSI %d", res.getDevice().getName(), res.getRssi()),
                            Toast.LENGTH_SHORT
                    ).show();
                }
            }

            @Override
            public void onScanFailed(int errorCode) {
                super.onScanFailed(errorCode);
                Toast.makeText(PluginActivity.this, "Scan is bulokens!", Toast.LENGTH_SHORT).show();
            }
        };
        List<ScanFilter> filters = new ArrayList<>();
        filters.add(new ScanFilter.Builder().setDeviceName("KunwareBLE1").build());
        scanner.startScan(filters, settings, scanCallback);
    }
}
```
My gradle file
```
plugins {
    id 'com.android.library'
}

android {
    namespace 'com.khynsoft.ble'
    compileSdk 32

    defaultConfig {
        minSdk 21
        targetSdk 32

        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
        consumerProguardFiles "consumer-rules.pro"
    }

    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
}

dependencies {

    compileOnly files('libs/classes.jar')
    implementation 'androidx.appcompat:appcompat:1.5.1'
    implementation 'com.google.android.material:material:1.7.0'
    testImplementation 'junit:junit:4.13.2'
    androidTestImplementation 'androidx.test.ext:junit:1.1.3'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.4.0'

    debugImplementation 'no.nordicsemi.android.support.v18:scanner:1.6.0'

}

```
I am making an android library and I am new to this.

philips77, 2022-11-08:
Yes, it may work in Android Studio, but then your build configuration may not include the library to the APK and it fails in runtime.

For example, you have
```groovy
debugImplementation 'no.nordicsemi.android.support.v18:scanner:1.6.0'
``
and it should be 
```groovy
implementation 'no.nordicsemi.android.support.v18:scanner:1.6.0'
```

closed 2022-11-08 by khynantoque