nordicsemi/Android-Scanner-Compat-Library #130: Hardware batching occassionally returns no results

albertmoravec, 2022-11-11:
I am currently trying to implement low-power background scanning using hardware batching, but on my device I sometimes get 0 `ScanResult`s back after starting particular scan. Restarting the scan with identical settings fixes the issue and I get results normally.

Here's how I configure and start the scan:
```kotlin
val scanSettings = ScanSettings.Builder()
  .setUseHardwareFilteringIfSupported(true)
  .setUseHardwareBatchingIfSupported(true)
  .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
  .setReportDelay(10 * 1000)
  .build()

val macFilter = ScanFilter.Builder()
  .setManufacturerData(
    0xFFFF,
    ubyteArrayOf(49u, 53u, 57u, 54u).toByteArray(),
    ubyteArrayOf(0xFFu, 0xFFu, 0xFFu, 0xFFu).toByteArray()
  )
  .build()

scanner.startScan(mutableListOf(macFilter), scanSettings, callback)
```

I am also able to reproduce the issue without using any scan filter, so it is not an issue in filter configuration. The issue is present with both `SCAN_MODE_BALANCED` and `SCAN_MODE_LOW_POWER`. 

Full example can be seen [here](https://gist.github.com/albertmoravec/257cbfde777d914c671ff040d67a764e).

Device used:
Model: Xiaomi Mi 9T Pro
Android: 11

I personally think this issue is going to be hardware-related, but I'm looking for any possible clues.

philips77, 2022-11-19:
Android has a limitation of only few scans per 30 second. If you start and stop scanning number of times in short period of time it will not return and scan results until you wait a bit. 
Can it be related?

albertmoravec, 2022-11-19:
Cannot rule it out entirely, but it's very unlikely, as the scanner got registered successfully (the `onScannerRegistered()` log message would indicate `status=6` and `scannerId=-1` otherwise AFAIK).