nordicsemi/Android-Scanner-Compat-Library #133: (Long Range (Bluetooth 5.2)) LE Coded PHY

orhangulcan, 2023-02-09:
Hi, 

I am making an android application that receives the advertises and the data (hex) in it using Bluetooth LE (5.2) Long Range (LE Coded). I can see LE Coded broadcasts using the nRF Connect application, but I cannot see them when I scan using this library in my own application. Do I need to add something to a ScanSettings to get the LE Coded Phys if I'm using the right library? or Android-BLE-Library should I use this library?

My phone is Samsung Galaxy S22 Ultra 5G, as far as I can see from its documentation, it uses a 5.2 supported chip.

```
            val settings: ScanSettings =
                ScanSettings.Builder()
                    .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
                    .setLegacy(false)
                    .setPhy(ScanSettings.PHY_LE_ALL_SUPPORTED)
                    .setNumOfMatches(ScanSettings.MATCH_NUM_MAX_ADVERTISEMENT)
                    .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                    .setMatchMode(ScanSettings.MATCH_MODE_AGGRESSIVE)
                    .setReportDelay(1000)
                    .build()

```
nRF Connect ScreenShot:
https://i.hizliresim.com/jyf3vrm.png

philips77, 2023-02-10:
Try changing `.setReportDelay(1000)` to `.setReportDelay(0)`.

philips77, 2023-02-10:
If your app is relying on Coded PHY, which is supported from Android 8, you don't have to use Scanner Compat Library at all, just the native `BluetoothLeScanner`. The point of scanner compat lib is to bring the newer API to Android 4.3 and 4.4.x.

The Android BLE Library is for handling connections, not scanning. If you want to connect to that device, you may find it useful, but it has nothing to do with scanning.

orhangulcan, 2023-02-11:
More clear now. Thanks for your answer.

closed 2023-02-13 by orhangulcan