nordicsemi/Android-Scanner-Compat-Library #134: setUseHardwareBatchingIfSupported

hjjice, 2023-02-21:
This interface must be set to true to scan my transparent transmission module，why?

closed 2023-02-21 by philips77

hjjice, 2023-09-19:
But I found that OPPO and OnePlus can connect when setUseHardwareBatchingIfSupported(false) is set. If set to true, the device cannot be connected.

hjjice, 2023-09-19:
但是   我这样操作之后   居然就可以连接了   实在不懂了：
settings
                    .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                    .setReportDelay(0)
                scanner.startScan(null, settings.build(), scanCallback)
                scannerState?.scanningStarted()
                Thread.sleep(1500)
                stopScan()
                Thread.sleep(1000)
                settings.setReportDelay(100)
                scanner.startScan(null, settings.build(), scanCallback)
                scannerState?.scanningStarted()

hjjice, 2023-09-19:
But after I did this, I was able to connect. I really don’t understand:
settings
.setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
.setReportDelay(0)
scanner.startScan(null, settings.build(), scanCallback)
scannerState?.scanningStarted()
Thread.sleep(1500)
stopScan()
Thread.sleep(1000)
settings.setReportDelay(100)
scanner.startScan(null, settings.build(), scanCallback)
scannerState?.scanningStarted()

philips77, 2023-09-26:
It may have a relation with how the Android caches the address type. 
As you may know, the Bluetooth address is 49 bit long. 49, not 48? Yes. The 49th bit is the address type, which may be *private* or *public*. Public addresses are registered by IEEE and accessible on their [website](https://regauth.standards.ieee.org/standards-ra-web/pub/view.html).

More about this here: https://novelbits.io/bluetooth-address-privacy-ble/

However, up until API 33 this last bit was not given explicitly to the API. This has changed with the introduction of [`getRemoteLeDevice(address, type)`](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getRemoteLeDevice(java.lang.String,%20int)) method and now in *androidx.bluetooth* library, which also has minSdk=33.

Until API 32 the address type had to be scanned and was assumed *public* otherwise, although most devices use random addresses. When Android scanned a packet from a new MAC address it was storing its type in a local database and when the user requested connection to a `BluetoothDevice` containing a 48-bit address, it was taking the type from there. That's why it's not possible to connect to a previously scanned device after toggling Bluetooth off and on. The database gets cleared and again a public address is assumed. The address type of bonded devices is stored permanently.

A *Connect Indication* request also contains 49-bit long address, so trying to connect to a valid 48-bit long address with a wrong type flag will fail.

Perhaps, scanning with a delay, which is offloaded to the controller, does not update the local database, hence Android awaits an advertising packet form a public address before sending Connection Indication. Only when the report delay is 0 and the packet is scanned  the list gets updated and following connections work. That is my assumption.

Also, we found that the `setReportDelay(..)` works best with delays 5 seconds or more, not 100 ms. Only then we were able to get true intervals or given length.