nordicsemi/Android-Scanner-Compat-Library #140: Could not find no.nordicsemi.android:thingylib:2.0.0.

asapsonter, 2023-06-28:
// nordic API
    implementation 'no.nordicsemi.android:ble:2.6.1'
    implementation 'no.nordicsemi.android.support.v18:scanner:1.6.0'
    implementation 'no.nordicsemi.android:thingylib:2.0.0'
    implementation 'no.nordicsemi.android:ble-livedata:2.6.1'

I have all this in my app level.

_ have removed  jcenter()_

asapsonter, 2023-06-28:
But I am unable to sync   'no.nordicsemi.android:thingylib:2.0.0'

roshanrajaratnam, 2023-06-28:
Hi what are you trying to achieve? Thingy is not available on mavenCentral as of now. You could include the project locally. Please refer to the sample [app](https://github.com/NordicSemiconductor/Android-Nordic-Thingy)

closed 2023-06-28 by roshanrajaratnam