nordicsemi/Android-Scanner-Compat-Library #146: Scan for nearby deivce with onBatchScanResults results list is empty

talfsaverone, 2024-06-04:
When scanning for the nearby devices I saw the following issue:

The `onBatchScanResults` returns an empty list while the `onScanResult` returns results.

My configuration is:
```kotlin
ScanSettings.Builder()
        .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
        .setReportDelay(5000L)
        .setUseHardwareBatchingIfSupported(false)
        .setUseHardwareFilteringIfSupported(false)
        .setLegacy(false)
        .build()
```

philips77, 2024-06-04:
Hmm.. intersting. What if you set `setUseHardwareBatchingIfSupported(true)`?

talfsaverone, 2024-06-05:
We saw from the comments in the code that it might have issues in some older devices so we haven't used it in our project but we were able to reproduce it with true as well.

Although it is tricky to reproduce this issue our product requires it to work 100% of the time, so what could be our alternative?