nordicsemi/Android-Scanner-Compat-Library #148: BluetoothLeScannerCompat.startScan fails on Google pixel 8 with android 14 and API 34-ext12

JakobTO2matic, 2024-08-21:
When I start scanning for bluetooth devices using BluetoothLeScannerCompat, the scan immedietly stops with error code 2 SCAN_FAILED_APPLICATION_REGISTRATION_FAILED.  This does not happen when using a Samsung phone, tablet (S22, S23, A9) or Oneplus phones. The scan is only started once, so i dont think it is an abundance of scans that is the issue, but i could be wrong.

Code for starting the scan:
```
        BluetoothLeScannerCompat scanner = getScanner();
        try {
            if(scanner==null){
                return false;
            }
            Log.e(TAG,"start scanner now!");
            scanner.startScan(null,ScanSettings,ScanCallback);
            ScannerStatus=ScanStatus.NO_ERROR;
            ScannerListeners.ScanStatus(ScanStatus.NO_ERROR,-1);
            result=future.get(30, TimeUnit.SECONDS);

        }
        catch (Exception e){
            e.printStackTrace();
        }
        finally {
            Log.e(TAG,"finally stop scanner now!");
            if(scanner!=null)
                scanner.stopScan(ScanCallback);
            ScanLock.unlock();
        }

        return result;
    }
```

Permissions asked for during runtime:

```
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
                String[] perms = { BLUETOOTH_CONNECT, BLUETOOTH_SCAN};
                EasyPermissions.requestPermissions(activity, context.getString(R.string.LocationPermission), 158, perms);
            }
            else{
                String[] perms = {Manifest.permission.ACCESS_FINE_LOCATION, Manifest.permission.BLUETOOTH, Manifest.permission.BLUETOOTH_ADMIN};
                EasyPermissions.requestPermissions(activity, context.getString(R.string.LocationPermission), 158, perms);
            }
```

```
    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />

    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"
        tools:targetApi="s" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-feature android:name="android.hardware.bluetooth" android:required="true"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30"
        />
```

I have tried to use `ACCESS_FINE_LOCATION` with `BLUETOOTH_SCAN` while removing `android:usesPermissionFlags="neverForLocation"`, but it does not resolve the issue. 

Logcat log:
```
11:32:23.212 BleScanner            com....oxygentherapy  E  Top start scan
11:32:23.212 BleScanner            com....oxygentherapy  E  Is Scan lock locked?:false
11:32:23.213 BleScanner            com....oxygentherapy  E  start scanner now!
11:32:23.215 BluetoothAdapter      com....oxygentherapy  D  isLeEnabled(): ON
11:32:23.218 BluetoothLeScanner    com....oxygentherapy  D  onScannerRegistered() - status=128 scannerId=0 mScannerId=0
11:32:23.219 ScannerListener       com....oxygentherapy  E  Scan status:NO_ERROR error code:-1
11:32:23.221 PlayCore              com....oxygentherapy  I  UID: [10394]  PID: [13295] OnRequestInstallCallback : onRequestInfo
11:32:23.221 PlayCore              com....oxygentherapy  I  UID: [10394]  PID: [13295] AppUpdateService : Unbind from service.
11:32:23.225 CheckUpdate           com....oxygentherapy  D  checked for update failed: -10: Install Error(-10): The app is not owned by any user on this device. An app is "owned" if it has been acquired from Play. (https://developer.android.com/reference/com/google/android/play/core/install/model/InstallErrorCode#ERROR_APP_NOT_OWNED)
11:32:23.225 ScannerListener       com....oxygentherapy  E  Scan status:SCAN_FAILED error code:2
11:32:23.225 BleScanner            com....oxygentherapy  E  finally stop scanner now!
```

philips77, 2024-09-27:
Hello,
Sorry for the long delay, I was busy with other projects.

Did you try restarting the phone? The error 128, which you get in onServiceRegistered is perhaps `NO_RESOURCES`:
https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/5738f83aeb59361a0a2eda2460113f6dc9194271/stack/include/gatt_api.h#49
Your code looks OKish and should work, if works on other phones.

JakobTO2matic, 2024-09-30:
Yes, I have tried to restart the device, but I will give it an extra try and report back.

Best regards,

Jakob Tørsløv
Product Developer
___________________________________

 [cid:86a9d2b7-f2c4-4341-834d-650d6875d5d8]

Mobil: +45  27 24 43 61
Mail: ***@***.***
Web: www.o2matic.com

O2matic ApS
Nørrelundvej 10
2730 Herlev
Denmark
Tlf: +45 50 52 98 10
[cid:9363199a-196f-4304-85eb-1ac789388373]

https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/5738f83aeb59361a0a2eda2460113f6dc9194271/stack/include/gatt_api.h#49